seb821/espRFLinkMQTT #1: Feature request - Wifi Captive Portal

agsochi, 2018-11-09:
Hi @seb821 
Thank you for grate firmware.
Can you please add Wifi Captive Portal like[ this one](https://github.com/tzapu/WiFiManager) to your project?

seb821, 2018-11-16:
Hi agsochi,
I thought about it, but it would add complexity to the code and require to spend more time on it.
Same for web configurable MQTT server : it is probably doable, but as I am not a c developer at all, every improvement requires quite a lot of time. So, I prefer to focus on other projects.
In fact, as I had spent quite some time on it, I just wanted to share my work as it could be useful to others, but I do not intend to make a firmware that will be maintained and improved regularly. 
Please be free to fork the project and improve it though.
Regards,

seb821, 2020-04-07:
Hey,

I added MQTT web configurable server in the dev branch. 

For wifi, I did several tests but I am not very satisfied at the moment
- [Wifimanager ](https://github.com/tzapu/WiFiManager) - it uses a lot of memory
-  [ESP_WiFiManager](https://github.com/khoih-prog/ESP_WiFiManager) - it works but still use quite some memory. It can be tested in the dev branch by defining USE_WM at the beginning of config. h
- [AutoConnect](https://hieromon.github.io/AutoConnect/) - very well documented but heavy. Sometimes the web page to add a network fails to appear. It can be tested in the dev branch by defining USE_AUTOCONNECT at the beginning of config. h

sylvaing, 2020-04-07:
nice improvement in sight !

seb821, 2020-04-10:
Dev branch now includes a WiFi management part. 
No use of external library in order to avoid additonnal memory useage or increase firmware size.
Sequence is: ESP starts access point after 1 min if WiFi connection failed, stays in this state for 5 min to give opportunity to update WiFi settings, then returns to normal WiFi connection attempt
Please feel free to test the dev branch and comment. 
This version is probably very close to being released.

closed 2020-04-30 by seb821