seb821/espRFLinkMQTT #3: rflink wifi board and hardware watchdog

brian-f, 2019-05-14:
Hi,
the official wifi daughter board from Nodo-shop has an hardware watchdog (ATtiny) which keeps resetting the nodemcu. 
[image]

In ESPeasy this watchdog is supported (WD), the value to set is i2C "38". Is it possible to add this to the code?

[https://github.com/letscontrolit/ESPEasy/blob/mega/src/Hardware.ino#L61-L81](url)

```
if (Settings.WDI2CAddress != 0)
  {
    delay(500);
    Wire.beginTransmission(Settings.WDI2CAddress);
    Wire.write(0x83);             // command to set pointer
    Wire.write(17);               // pointer value to status byte
    Wire.endTransmission();
    Wire.requestFrom(Settings.WDI2CAddress, (uint8_t)1);
    if (Wire.available())
    {
      byte status = Wire.read();
      if (status & 0x1)
      {
        String log = F("INIT : Reset by WD!");
        addLog(LOG_LEVEL_ERROR, log);
        lastBootCause = BOOT_CAUSE_EXT_WD;
      }
    }
  }

```
thanks

seb821, 2019-06-01:
This is a specific case so I will not implement it. But you are free to fork the code and add it if you want.

rverheijdengit, 2020-01-16:
Anyone found a solution for this or point me to code that has this implemented i now to have the rest issue because i'm using the original rflink wifi board

seb821, 2020-04-10:
@brian-f @rverheijdengit 
I am re-opening this issue. I have some time to try it. Problem is, without the hardware it is difficult to know whether it works.
If someone volunteers to test, I am ok to try to add the code as a compilation option.
Please let me know.

seb821, 2020-04-17:
Ok, this was tested and discussed: https://www.letscontrolit.com/forum/viewtopic.php?f=12&t=7495

I shall include it in the next revision as a compilation option.

seb821, 2020-04-30:
Implement in dev branch, will be available in master soon

closed 2020-04-30 by seb821