seb821/espRFLinkMQTT #4: Stops sending received messages after a day or so

francisp2, 2019-09-26:
It is working fine for a day or so, after that I don't receive rflink codes anymore (rflink/uptime keeps receiving). If I reset the esp from the webpage, it works again for a day or so.

seb821, 2019-10-11:
It is difficult to analyse with so little information. 
I would suggest to enable serial debug by uncomenting #define SERIAL_DEBUG in config.h. Of course, you would need to connect with a computer to monitor the messages when it stops working.
It could also be the MEGA / RFLink that stops transmitting messages to the ESP. Did you try to click on Reset Mega on the web interface ? Does it work ? If so, you can try to define a value for RESET_MEGA_INTERVAL that will reset the mega if the ESP stops receiving messages (10 * 1000 * 60 for 10 minutes for example). Of course, you have to wire the ESP pin defined by MEGA_RESET_PIN to MEGA reset pin. If it still does not work, you could try to wire the previous ESP in to ESP reset pin.

seb821, 2020-03-28:
Lot of optimization was done with the last commit. It may help with your issue.

closed 2020-03-28 by seb821