seb821/espRFLinkMQTT #6: Does not compile on latest "Generic ESP8266 Module" version

jrhbcn, 2020-02-27:
Hello,

When trying to compile the code in Arduino IDE with latest "Generic ESP8266 Module" version (2.6.3) the following error appears:

`
config.h:15:52: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(int, int, bool, int)'

 SoftwareSerial softSerial(4, 2, false,BUFFER_SIZE+2); // software serial RX from GPIO4/D2 pin (unused pin on ESP01), software serial TX to RFLink on GPIO2/D4 pin - uncomment to use software serial
...
     SoftwareSerial(int8_t rxPin, int8_t txPin = -1);

SoftwareSerial.h:94:5: note:   candidate expects 2 arguments, 4 provided

no matching function for call to 'SoftwareSerial::SoftwareSerial(int, int, bool, int)'` 

Using version 2.5.2 solves the issue.

TimothyGaray, 2020-03-07:
Thank you.  Just ran into that same message for a NodeMCU 1.0 (ESP-12E) module.  Rolling back to 2.5.2 resolved it for me as well.

seb821, 2020-03-10:
Last version is using RX/TX ESP pins by default.

For core 2.6.3, you need to remove last argument and use this: "SoftwareSerial softSerial(4, 2, false)" instead of "SoftwareSerial softSerial(4, 2, false,BUFFER_SIZE+2)"

closed 2020-03-10 by seb821