seb821/espRFLinkMQTT #7: Feature Request: Ability to Set Retain on Published Messages

TimothyGaray, 2020-03-19:
A feature that I haven't easily been able to figure out how to add myself to the sketch would be to set the MQTT published message with the retain=1.

Case: door/window switches send code when their state changes such as the door opens.  If home home automation software (Home Assistant) is not running or is rebooting at the time then it will miss that message and think the door or window is closed when it is actually open.

Setting the message to be retained will keep it available for when HA comes back up.

I'm not sure the impact for all published messages being retained or if it would really matter.  Ideally the ability to specify which IDs get retained messages would be nice.

Thank you.  Aside from some little tweaks, this has been working nicely for me.

-TimG

seb821, 2020-03-28:
Hey,

See [PubSubClient documentation](https://pubsubclient.knolleary.net/api.html).

Try to replace
> MQTTClient.publish(MQTT_TOPIC,JSON);

with

> MQTTClient.publish(MQTT_TOPIC,JSON,1);

I will keep the issue open for now and think about adding a setting for that.

TimothyGaray, 2020-04-02:
Thank you, I will give that a try.

seb821, 2020-04-07:
Implemented in the dev branch