seb821/espRFLinkMQTT #8: Feature Request: Move ID to JSON Payload

TimothyGaray, 2020-03-19:
Another thing I've seen with my two-code door switch is that it sends a message with one ID for open and a message with a different ID for closed.

espRFLinkMQTT turns these into two different topics:

open: rflink/SelectPlus-065afa
close: rflink/SelectPlus-065af8

With the same payload for both.

{"SWITCH":"02","CMD":"ON","SWITCH02":"ON","CHIME":"01"}

It might help in some cases if there was the option to keep the topic at just the name (rflink/SelectPlus) and move the ID down as an item in the JSON payload {"ID":"065afa"}

Thank you!

-TimG

seb821, 2020-03-28:
Hey,

Well, this is not really nice from the device itself.

Problem is :
- If I apply what you ask to every device, then two completely different devices using the same protocol will end up being reported together.

- If the option is to apply only on some specific devices, it requires adding some code in the configuration and it will complexity and use more memory.

So, I would say it is probably better to handle these specific cases on the home automation side.

TimothyGaray, 2020-04-02:
Thank you, I understand.  I have been able to do the workaround in Home Assistant.

seb821, 2020-04-07:
Good to here that.

After some thinking about that, I will not implement such fonctionnality as it is really specific.

closed 2020-04-07 by seb821