seb821/espRFLinkMQTT #10: pico change for softserial

sylvaing, 2020-03-30:
Hi @seb821, I'm sylvain from jeedom old forum.
I upgrad my two esprflinkMQTT, but I use Wemose D1 mini, and soft serial to change TX pin.
I Think that you must reverse 3 lines in you code to better understand:

You write:
//SoftwareSerial& rflinkSerialTX = softSerial;
//SoftwareSerial softSerial(4, 2, false);
//#include <SoftwareSerial.h>

But it's (i think):
//#include <SoftwareSerial.h>
//SoftwareSerial softSerial(4, 2, false);
//SoftwareSerial& rflinkSerialTX = softSerial;

PS: Good features in this version !

seb821, 2020-03-30:
Yes indeed. I changed to RX/TX for default settings - as it is the most commonly used - and tried to gather the other lines. I shall keep the issue open and fix this next time I update the software.

Good if you like the new features. This is actually quite usefull to avoid frequent firmware compilation and upload just for one device changing its ID.

seb821, 2020-04-07:
Change done in the dev branch

closed 2020-04-07 by seb821