seb821/espRFLinkMQTT #11: Does it store the json messages in local file somewhere on device?

happytm, 2020-04-29:
Hi,

Thank you for sharing your very useful code. I would like to know if json data is stored in SPIFFS on the device? If so roughly how many messages can be stored with 3 MB Spiffs partition?

Thanks.

seb821, 2020-04-30:
Hi,

No. SPIFFS is not used at all. Only the last json message is saved and it is done in memory, which means it is lost on reboot. 

espRFLink acts as a gateway. Last json is only used to compare information and avoid publishing too frequently the same information on the MQTT server (this behavior can be adjusted for each ID with "Publish Interval" setting and disabled by selecting 0).

happytm, 2020-04-30:
Understood. Thank you.

closed 2020-05-01 by seb821