seb821/espRFLinkMQTT #12: Need your help with MQTT

happytm, 2020-05-04:
I would like to use your code for web UI it offers. I do not have any RFLink  devices but I get some MQTT messages from several devices onlocal network which are in following format:

>received topic 'DeviceStatus/{"Location":"Bedroom1","SingnalStrength":-58,"DeviceMode":0,"DeviceIP":6,"WiFiChannel":7,"SleepTime":1,"UpTime":130}
received topic 'SensorValues/{"Location":"Kitchen","Volts":2.62,"Sensor1":"Temperature","SensorValue1":1,"Sensor2":"Humidity","SensorValue2":2,"Sensor3":"Pressure","SensorValue3":244,"Sensor4":"Light","SensorValue4":81}
received topic 'Warning/Battery Low/Kitchen
received topic 'DeviceStatus/{"Location":"Kitchen","SingnalStrength":-52,"DeviceMode":0,"DeviceIP":16,"WiFiChannel":7,"SleepTime":1,"UpTime":139}
received topic 'Warning/Battery Low/Livingroom

Is there any possibility to strip all code for RFLink and use your Web server to display these MQTT messages? I tried to do that but I broke to many things in code so it does not compile. Please give me some hints if possible. 

Thanks.

seb821, 2020-05-14:
See norf branch for a stripped version

closed 2020-05-14 by seb821

happytm, 2020-05-14:
@seb821  thank you so much. you saved me lot of time.