seb821/espRFLinkMQTT #15: MQTT masseges are not published, only in debug mode

J0nnyBr4v0, 2020-06-25:
HI seb821,

amazing job, love this piece of code. Thank you very much for your work.
using the following kit from nodo shop:
https://www.nodo-shop.nl/nl/rflink-gateway/127-rflink-arduino-antenne-.html
got everything up and running based on your instruction but unfotunately no sensor MQTT messages are published to my broker.
I can see everything on the webinterface see screenshot attached but if I subscribe to the topic rflink/#, can only see the online and uptime topics.
[image]
If I enable debug mode, all debug messegaes are beeing correctly published.

Sorry maybe I miss something, I'm still a noob ;)
Hope you can help.

System info:

Version | 20200328
-- | --
Uptime | 1 min
WiFi network | XXXXXXXXXXXXXXX (E0:28:6D:14:XX:XX)
WiFi RSSI | -52 dB
IP address (MAC) | 192.168.1.79 (CC:50:E3:2B:XX:XX)
ESP pin to reset MEGA | GPIO 0
MQTT server and port | 192.168.1.55:1883
MQTT debug | disabled
MQTT connection state ❔ | 0
MQTT topics | publish (json)rflink/Protocol_Name-IDcommands to rflinkrflink/cmdlast will ( 1 / 0 )rflink/onlineuptime (min, every 5)rflink/uptimedebug (data from rflink)rflink/debug | publish (json) | rflink/Protocol_Name-ID | commands to rflink | rflink/cmd | last will ( 1 / 0 ) | rflink/online | uptime (min, every 5) | rflink/uptime | debug (data from rflink) | rflink/debug
publish (json) | rflink/Protocol_Name-ID
commands to rflink | rflink/cmd
last will ( 1 / 0 ) | rflink/online
uptime (min, every 5) | rflink/uptime
debug (data from rflink) | rflink/debug
Free Mem | 19000
Free Stack | 3664
Heap Max Free Block | 18736
Heap Fragmentation | 1%

J0nnyBr4v0, 2020-06-26:
basically this topics are working:

commands to rflink | rflink/cmd
-- | --
last will ( 1 / 0 ) | rflink/online
uptime (min, every 5) | rflink/uptime
debug (data from rflink) | rflink/debug

only this topic is not working, even if its desplayed on the webinterface:

publish (json) | rflink/Protocol_Name-ID
-- | --

J0nnyBr4v0, 2020-06-27:
sorry was to stupid, I needed to add the device id. Now its working .
Thanks anyway

closed 2020-06-27 by J0nnyBr4v0

gbarocsi, 2021-01-11:
Hi!
In which commit is this fixed? I have the same issue now.