seb821/espRFLinkMQTT #17: Will you release v1 from dev branch

Minims, 2020-09-10:
Will you release v1 from dev branch or it's still not stable for now ?  Thanks.

seb821, 2021-04-23:
Yes, hopefully within a few days. Still would be interested that a few person test the current dev branch before releasing for good.

Minims, 2021-04-23:
I'm using it for months on my side. Seems OK for me.

J0nnyBr4v0, 2021-04-23:
Just updated to dev, works like a charm so far.
Would also be nice to be able to backup the complete filtering list.
And is there a way to extend the filtering to more then 32 devices?
Thanks anyway, great tool!

mgeers, 2021-04-23:
Updated to DEV as well. Working without hiccup. Thanks!
Quick question, though. What is the difference between the 2 bin files?

seb821, 2021-04-28:
> Would also be nice to be able to backup the complete filtering list.

I have added a few options in dev branch : 
```
/idfiltering_config_h => code to update config.h
/idfiltering_restore_curl => restore with curl command
/idfiltering_restore_url => restore with url in browser
/configuration.json => save configuration in json file (cannot be imported)
```
Do not hesitate to test and provide feedback.

Please note that it would be safer to backup your configuration before trying this (copy paste or print in pdf).  Indeed, current CONFIG_VERSION is 20210418. If it differs from your current system, configuration is automatically erased. Though, eeprom structure was not changed recently, so if you compile with a CONFIG_VERSION matching your current installation and upload the firmware, you may be able to update without erasing.

> And is there a way to extend the filtering to more then 32 devices?

Yes indeed. You would have to compile your own firmware and change FILTERED_ID_SIZE in espRFLinkMQTT.h file. I had done tests with 64 devices. Above, you will get issues with web interface and/or run out of space in eeprom. As this affects eeprom size, it will also erase your current configuration.

> Thanks anyway, great tool!

Thanks

seb821, 2021-04-28:
> Quick question, though. What is the difference between the 2 bin files?

The bin file ending by "rflink-wifi-board" is enabling a watchdog required for "official" [RFLink Wifi board](https://www.nodo-shop.nl/en/home/191-rflink-wifi-koppelprint.html)

seb821, 2021-05-01:
Release v1.0 just published

closed 2021-05-01 by seb821