seb821/espRFLinkMQTT #22: Serial to TCP bridge

bilbolodz, 2021-02-11:
Could you please consider adding "serial to TCP bridge" option to your project? Unfortunately Domoticz doesn't support RFLink via MQTT. Home assistant support only one RFLink over TCP connection, more than one RFLinks only through MQTT.
There are project which implements "serial to TCP bridge" (e.x. Tasmota or esp-link) but non of them has both features together.
I'd like to connect 3 my RFLinks (different frequencies) to Domoticz and Homeassistant at the same time.

TD-er, 2021-02-11:
What about ESPEasy using the Ser2Net plugin?
Initially it was written for the RFLink.

bilbolodz, 2021-02-11:
ESPEasy Ser2Net support only one TCP connection (connecting second disconnects first) and doesn't support MQTT bridge.

TD-er, 2021-02-11:
Valid points.
Just wanted to make sure you at least considered it and not "simply unaware of it".

sylvaing, 2021-02-11:
I use 2 rflink with this project on home assistant. You must just configure your rflink with two different mqtt topic, and go

bilbolodz, 2021-02-11:
I know but there is no way to use it with Domoticz at the same time.

bilbolodz, 2021-02-13:
I've started to implement these issue myself. I hope it will be ready soon.

TD-er, 2021-02-13:
> ESPEasy Ser2Net support only one TCP connection (connecting second disconnects first) and doesn't support MQTT bridge.
> And for me it's not solution because I need TCP connection (for Domoticz) and MQTT (for multiple RFLinks in HA) at the same time.

Just FIY, with [this PR](https://github.com/letscontrolit/ESPEasy/pull/3483) part of your mentioned shortcomings should be tackled.

bilbolodz, 2021-02-19:
I've working version with "TCP server" (multiple connection). I will clean up some things in code and I'll publish it. If someone is interested in testing I can share "beta version".

seb821, 2021-04-23:
No time to work on such a functionality but feel free to share your work.

closed 2021-04-23 by seb821

matthewjporter, 2021-12-06:
@sylvaing 

> I use 2 rflink with this project on home assistant. You must just configure your rflink with two different mqtt topic, and go

I am trying to setup a 443 and 868 rflink. Can you share your config and how you got the devices into HA as mine do not seem to auto add