seb821/espRFLinkMQTT #23: Invalid json on publish

HenJou, 2021-04-20:
Hi I tested the the espRFLinkMQTT and i tryed the latest dev one but I get an error on my DKW2012 Weather station on the json that is published to my mqtt server .  This is the json that is returned

{"TEMP":15.8,"HUM":48,"WINSP":0.0,"WINGS":0.0,"RAIN":5.4,"WINDIR":0010,"BAT":"OK"}

if I run this thru json validate at https://jsonlint.com/ I get thats WINDIR value that's not valid.

My Milight is working and also my other Temp sensors.

[image]

HenJou, 2021-04-20:
What I got from rflink is that the WINDIR of the DKW2012 is a value from 0-15 and every 1 is 22.5 deg. so the 0010 above is then 10 * 22.5 = 225 degrees.  Hope this can help.

seb821, 2021-04-22:
I have just pushed a fix on the dev branch to convert it to a proper integer (without the zeros before).
Though, I did not implement the multiplication by 22.5 as RFLink reference says it is an integer (though it "reflects" an angle).

HenJou, 2021-04-25:
Thank seb821 I test the dev and i do get a valid json back and I got the value back to Home assistant with the following configuration.yaml under sensors: , so no need to add the conversion also.
  - platform: mqtt
    name: "DKW Wind Direction"
    state_topic: "rflink/DKW2012-weather"
    value_template: "{{ value_json['WINDIR'] *22.5  }}"

closed 2021-04-25 by HenJou