seb821/espRFLinkMQTT #30: Connection to Home assistant

matthewjporter, 2021-12-06:
Hi, Not sure if you can help. Tyring to connect with Home Assistant

I can see my devices are being received by espRFLinkMQTT on the live data screen:

It is a Thermostat with a receiver switch that simply turns the relay on/off

This is Off
20;90;X2D;ID=0011818;SWITCH=00;CMD=ON;RC=00fe;S=0c;EXT=Tybox;BAT=OK; | rflink/X2D-0011818 | {"SWITCH":"00","CMD":"ON","SWITCH00":"ON","RC":00fe,"S":0c,"EXT":Tybox,"BAT":"OK"}

This is On
20;8F;X2D;ID=0011818;SWITCH=00;CMD=ON;RC=1cfe;S=0c;EXT=Tybox;BAT=OK; | rflink/X2D-0011818 | {"SWITCH":"00","CMD":"ON","SWITCH00":"ON","RC":1cfe,"S":0c,"EXT":Tybox,"BAT":"OK"}

If i use MQTTLens I can see the topics being created on my mqtt server but what I am struggling with is that I cannot create/see the device Home Assistant

Any assistance would be appreciated

bakko65, 2021-12-06:
Hi, do you see nothing in the log of HA also?
And in the mqtt integration log?
If not, then the topic is not received by HA.
Do you use mqtt for other devices in HA or is this you're first device?

matthewjporter, 2021-12-07:
Yep, If I set HA MQTT integration to Listen on rflink/# I can see the topic come in and then out when I trigger my switch in HA but it never gets to espRFLinkMQTT from HA

bakko65, 2021-12-07:
For my cover I send: 10;RTS;8833FF;0;UP
In the HA log there is a warning because this is not supported but it works nicely.
You need to send with 10;
Something like 10;ID=0011818;SWITCH=00;CMD=ON
Did you try/do that?